adafruit/Adafruit_CircuitPython_BMP280 #1: review

ladyada, 2017-10-24:
heya woo its my first driver for the bundle, looks good to y'all? now is a good time to tell me if something needs changing :) will use this as a skeleton for the other drivers i will write

dhalbert, 2017-10-24:
Great! If you make this kind of thing be a pull request it's easy for us to add comments to individual lines of code.

Prefix all the constants with an underscore unless you want them to be available at run-time to the library user. E.g., do:
`_BMP280_ADDRESS = const(0x77)`
The underscore makes the name private to the module or class, and makes it not available in the module or class dict in the compiled code. This saves a noticeable amount of space when there a lot of constants because the strings for the names are no longer in the compiled code.

See http://circuitpython.readthedocs.io/en/2.x/docs/library/micropython.html?highlight=underscore#micropython.const:
>[...] if a constant begins with an underscore then it is hidden, it is not available as a global variable, and does not take up any memory during execution.

The code itself is nice and clean and looks good (without testing - I don't have a BMP280 to try).

tannewt, 2017-10-24:
Overall, it looks really good! I love all of the properties. It makes it so simple to use.

A couple things I see:

* I setup up the ReadTheDocs [here](https://circuitpython.readthedocs.io/projects/bmp280/en/latest/) and the API docs aren't getting built. When looking at the [build output](https://readthedocs.org/projects/adafruit-circuitpython-bmp280/builds/6171058/) its because const is unknown. To fix it you'll need to add  `from micropython import const` to the driver file and `autodoc_mock_imports = ['micropython']` to conf.py for Sphinx.
* Nitpicky but the comments on properties should be phrased like they are variables not functions because from the outside they look like variables. So, "Gets the compensated temperature in degrees Celsius." becomes "The compensated temperature in degrees Celsius."

ladyada, 2017-10-27:
ok fixed!
https://github.com/adafruit/Adafruit_CircuitPython_BMP280/commit/060cb391beb1a89e3ff4f375ef47745e95dcce7e

ill close this and submit to the bundle?

dhalbert, 2017-10-27:
>ill close this and submit to the bundle?
Sure!

closed 2017-10-27 by ladyada