adafruit/Adafruit_CircuitPython_BMP280 #12: Problem with Pi 3 model B+ & BMP280

GK8591, 2019-02-18:
Hello world, I got this error after install setup and running <bmp280_simpletest.py>:

Traceback (most recent call last):
  File "bmp280_simpletest.py", line 3, in <module>
    import board
  File "build/bdist.linux-armv7l/egg/board.py", line 32, in <module>
  File "build/bdist.linux-armv7l/egg/adafruit_blinka/agnostic/__init__.py", line 21, in <module>
AttributeError: 'module' object has no attribute 'implementation'

Anyone with an idea to fix it? Thx in advance 👍

ladyada, 2019-02-18:
are you using python 3?

GK8591, 2019-02-18:
pi@raspberrypi:~/SunFounder_SensorKit_for_RPi2/Python/Adafruit_Python_DHT/exampl
es/Adafruit_CircuitPython_BMP280 $ sudo python setup.py install
/usr/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution opti                                     on: 'long_description_content_type'
  warnings.warn(msg)
running install
running bdist_egg
running egg_info
creating adafruit_circuitpython_bmp280.egg-info
writing requirements to adafruit_circuitpython_bmp280.egg-info/requires.txt
writing adafruit_circuitpython_bmp280.egg-info/PKG-INFO
writing top-level names to adafruit_circuitpython_bmp280.egg-info/top_level.txt
writing dependency_links to adafruit_circuitpython_bmp280.egg-info/dependency_li                                     nks.txt
writing manifest file 'adafruit_circuitpython_bmp280.egg-info/SOURCES.txt'
writing manifest file 'adafruit_circuitpython_bmp280.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv7l/egg
running install_lib
running build_py
creating build
creating build/lib.linux-armv7l-2.7
copying adafruit_bmp280.py -> build/lib.linux-armv7l-2.7
creating build/bdist.linux-armv7l
creating build/bdist.linux-armv7l/egg
copying build/lib.linux-armv7l-2.7/adafruit_bmp280.py -> build/bdist.linux-armv7                                     l/egg
byte-compiling build/bdist.linux-armv7l/egg/adafruit_bmp280.py to adafruit_bmp28                                     0.pyc
creating build/bdist.linux-armv7l/egg/EGG-INFO
copying adafruit_circuitpython_bmp280.egg-info/PKG-INFO -> build/bdist.linux-arm                                     v7l/egg/EGG-INFO
copying adafruit_circuitpython_bmp280.egg-info/SOURCES.txt -> build/bdist.linux-                                     armv7l/egg/EGG-INFO
copying adafruit_circuitpython_bmp280.egg-info/dependency_links.txt -> build/bdi                                     st.linux-armv7l/egg/EGG-INFO
copying adafruit_circuitpython_bmp280.egg-info/requires.txt -> build/bdist.linux                                     -armv7l/egg/EGG-INFO
copying adafruit_circuitpython_bmp280.egg-info/top_level.txt -> build/bdist.linu                                     x-armv7l/egg/EGG-INFO
zip_safe flag not set; analyzing archive contents...
creating dist
creating 'dist/adafruit_circuitpython_bmp280-3.0.10-py2.7.egg' and adding 'build                                     /bdist.linux-armv7l/egg' to it
removing 'build/bdist.linux-armv7l/egg' (and everything under it)
Processing adafruit_circuitpython_bmp280-3.0.10-py2.7.egg
Removing /usr/local/lib/python2.7/dist-packages/adafruit_circuitpython_bmp280-3.                                     0.10-py2.7.egg
Copying adafruit_circuitpython_bmp280-3.0.10-py2.7.egg to /usr/local/lib/python2                                     .7/dist-packages
adafruit-circuitpython-bmp280 3.0.10 is already the active version in easy-insta                                     ll.pth

Installed /usr/local/lib/python2.7/dist-packages/adafruit_circuitpython_bmp280-3                                     .0.10-py2.7.egg
Processing dependencies for adafruit-circuitpython-bmp280==3.0.10
Searching for adafruit-circuitpython-busdevice==2.2.9
Best match: adafruit-circuitpython-busdevice 2.2.9
Processing adafruit_circuitpython_busdevice-2.2.9-py2.7.egg
adafruit-circuitpython-busdevice 2.2.9 is already the active version in easy-ins                                     tall.pth

Using /usr/local/lib/python2.7/dist-packages/adafruit_circuitpython_busdevice-2.                                     2.9-py2.7.egg
Searching for Adafruit-Blinka==1.2.3
Best match: Adafruit-Blinka 1.2.3
Processing Adafruit_Blinka-1.2.3-py2.7.egg
Adafruit-Blinka 1.2.3 is already the active version in easy-install.pth

Using /usr/local/lib/python2.7/dist-packages/Adafruit_Blinka-1.2.3-py2.7.egg
Searching for sysv-ipc==1.0.0
Best match: sysv-ipc 1.0.0
Processing sysv_ipc-1.0.0-py2.7-linux-armv7l.egg
sysv-ipc 1.0.0 is already the active version in easy-install.pth

Using /usr/local/lib/python2.7/dist-packages/sysv_ipc-1.0.0-py2.7-linux-armv7l.e                                     gg
Searching for rpi-ws281x==4.1.0
Best match: rpi-ws281x 4.1.0
Processing rpi_ws281x-4.1.0-py2.7-linux-armv7l.egg
rpi-ws281x 4.1.0 is already the active version in easy-install.pth

Using /usr/local/lib/python2.7/dist-packages/rpi_ws281x-4.1.0-py2.7-linux-armv7l                                     .egg
Searching for RPi.GPIO==0.6.5
Best match: RPi.GPIO 0.6.5
Adding RPi.GPIO 0.6.5 to easy-install.pth file

Using /usr/lib/python2.7/dist-packages
Searching for Adafruit-PureIO==0.2.3
Best match: Adafruit-PureIO 0.2.3
Processing Adafruit_PureIO-0.2.3-py2.7.egg
Adafruit-PureIO 0.2.3 is already the active version in easy-install.pth

Using /usr/local/lib/python2.7/dist-packages/Adafruit_PureIO-0.2.3-py2.7.egg
Searching for Adafruit-PlatformDetect==0.0.10
Best match: Adafruit-PlatformDetect 0.0.10
Processing Adafruit_PlatformDetect-0.0.10-py2.7.egg
Adafruit-PlatformDetect 0.0.10 is already the active version in easy-install.pth

Using /usr/local/lib/python2.7/dist-packages/Adafruit_PlatformDetect-0.0.10-py2.                                     7.egg
Finished processing dependencies for adafruit-circuitpython-bmp280==3.0.10

I'm using Python 2.7.13 (default, Sep 26 2018, 18:42:22)
[GCC 6.3.0 20170516] on linux2

ladyada, 2019-02-18:
did you purchase a "SunFounder_SensorKit"?

GK8591, 2019-02-18:
Yes, additional one, too.

ladyada, 2019-02-18:
ok, please contact SunFounder_SensorKit for support

closed 2019-02-18 by ladyada