adafruit/Adafruit_CircuitPython_BMP280 #16: 0x76 Address issue with I2c connection in Rpi3

vasee007, 2019-09-07:
Hi,
    I'm using a BMP280 sensor module with raspberry pi 3 using I2c connection... after connection when I check for the device availability using i2cdetect -y 1 command I get an address of 76 in the console for the device but the library takes the default address as 77 how to change the address manually whenever required???.... I'm using this kind of sensor module(image attached below)

[image]

ladyada, 2019-09-07:
https://github.com/adafruit/Adafruit_CircuitPython_BMP280/blob/master/adafruit_bmp280.py#L374

closed 2019-09-07 by ladyada

vasee007, 2019-09-07:
its mentioned  "Driver for I2C connected BMP280. Default address is 0x77 but another address can be passed in as an argument"

how to do that??

vasee007, 2019-09-07:
when I run this code :

import time

import board
# import digitalio # For use with SPI
import busio

import adafruit_bmp280

# Create library object using our Bus I2C port
i2c = busio.I2C(board.SCL, board.SDA)
bmp280 = adafruit_bmp280.Adafruit_BMP280_I2C(i2c)

# OR create library object using our Bus SPI port
#spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
#bmp_cs = digitalio.DigitalInOut(board.D10)
#bmp280 = adafruit_bmp280.Adafruit_BMP280_SPI(spi, bmp_cs)

# change this to match the location's pressure (hPa) at sea level
bmp280.sea_level_pressure = 1013.25

while True:
    print("\nTemperature: %0.1f C" % bmp280.temperature)
    print("Pressure: %0.1f hPa" % bmp280.pressure)
    print("Altitude = %0.2f meters" % bmp280.altitude)
    time.sleep(2)

jerryneedell, 2019-09-07:
use 
```
bmp280 = adafruit_bmp280.Adafruit_BMP280_I2C(i2c, address = 0x76)
```

LuckyLyz, 2021-02-26:
> use
> 
> ```
> bmp280 = adafruit_bmp280.Adafruit_BMP280_I2C(i2c, address = 0x76)
> ```
RuntimeError: Failed to find BMP280! Chip ID 0x60

jerryneedell, 2021-02-26:
Are you using the Adafruit BMP280 Breakout board? If not, what board are you using? The default I2C address is 0x77 or it can be changed to 0x76 if the SDO pin is Grounded.

garciajg, 2021-12-13:
I'm running into this issue with Raspberry Pi 4