adafruit/Adafruit_CircuitPython_BMP280 #23: maximum recursion depth exceeded

hukuzatuna, 2020-06-12:
I am running the BMP280 on a Feather STM32F405 Express running the latest firmware (2020-06-12 downloaded around 0645 Pacific) and library bundle. When I run the example code from this repo, it prints the temperature and pressure once, then throws a "maximum recursion depth exceeded" error.
```
Temperature: 21.1 C
Pressure: 1005.7 hPa
Traceback (most recent call last):
  File "code.py", line 25, in <module>
  File "adafruit_bmp280.py", line 358, in altitude
  File "adafruit_bmp280.py", line 328, in pressure
  File "adafruit_bmp280.py", line 149, in _read_temperature
  File "adafruit_bmp280.py", line 214, in mode
  File "adafruit_bmp280.py", line 180, in _write_ctrl_meas
RuntimeError: maximum recursion depth exceeded
```

I'm not sure the proper way to debug this. The code runs fine on the Feather M4 Express. Let me know what I can do to help debug this!

tannewt, 2020-06-12:
I believe this is an issue with the stack size setting on the F405. `maximum recursion depth exceeded` is also means it ran out of stack space. @hierophect would know for sure.

hierophect, 2020-06-12:
Hmm. I thought I fixed this a while ago. @hukuzatuna, can you post the actual version of Circuitpython you are using as reported by the REPL? Should be the first line that shows up on reset or a manual soft-reboot (ctrl-c, ctrl-d)

jerryneedell, 2020-06-12:
I was able to run the bmp208_simpletest on v5.4.0-beta

hukuzatuna, 2020-06-12:
@hierophect It was 5.3.0 downloaded this morning. I'm not sure of the specific build since I already re-flashed with 5.4.0 Beta. 5.4.0 Beta works, by the way, so we can probably close this one.

closed 2020-06-30 by dherrada