adafruit/Adafruit_CircuitPython_BMP280 #28: Wrong _CPU_ID detected in adafruit_bmp280.py

feranick, 2021-04-02:
I have a Adafruit BMP280, operating via i2C. It's the only board connected to the Raspberry PI. When using the CircuitPython library, I get this error:

`RuntimeError: Failed to find BMP280! Chip ID 0x60`

The library (`adafruit_bmp280.py`) expects to have the `CPU_ID = 0x058`, therefore triggering the error above. I am not sure why the value of the _CPU_ID is hardcoded, but the crash can be fixed by changing the `_CPU_ID` value in `adafruit_bmp280.py` to `0x60`. With that change, the library works fine.

With this bug report, I wonder whether it's needed to have a hardcoded value for `_CPU_ID` or if it is, whether adding values corresponding to official Adafruit BMP280 board may be needed.

dhalbert, 2021-04-02:
A BME390 and a BME280 have a chip id of 0x60. Could you double-check what the labeling on your board?

feranick, 2021-04-02:
You are absolutely correct, it's a BME280. Sorry for the trouble.

closed 2021-04-02 by feranick