adafruit/Adafruit_CircuitPython_BMP280 #32: Missing Type Annotations

FoamyGuy, 2021-09-27:
There are missing type annotations for some functions in this library.

The `typing` module does not exist on CircuitPython devices so the import needs to be wrapped in try/except to catch the error for missing import. There is an example of how that is done here:

```python
try:
    from typing import List, Tuple
except ImportError:
    pass
```

Once imported the typing annotations for the argument type(s), and return type(s) can be added to the function signature. Here is an example of a function that has had this done already:

```python
def wrap_text_to_pixels(
    string: str, max_width: int, font=None, indent0: str = "", indent1: str = ""
) -> List[str]:
```
If you are new to Git or Github we have a guide about contributing to our projects here: https://learn.adafruit.com/contribute-to-circuitpython-with-git-and-github

There is also a guide that covers our CI utilities and how to run them locally to ensure they will pass in Github Actions here: https://learn.adafruit.com/creating-and-sharing-a-circuitpython-library/check-your-code In particular the pages: `Sharing docs on ReadTheDocs` and `Check your code with pre-commit` contain the tools to install and commands to run locally to run the checks.

If you are attempting to resolve this issue and need help, you can post a comment on this issue and tag both @foamyguy and @kattni or reach out to us on Discord: https://adafru.it/discord in the `#circuitpython-dev` channel.

The following locations are reported by mypy to be missing type annotations:

- [ ] adafruit_bmp280.py:210
- [ ] adafruit_bmp280.py:225
- [ ] adafruit_bmp280.py:242
- [ ] adafruit_bmp280.py:257
- [ ] adafruit_bmp280.py:272
- [ ] adafruit_bmp280.py:377
- [ ] adafruit_bmp280.py:381
- [ ] adafruit_bmp280.py:389
- [ ] adafruit_bmp280.py:393
- [ ] adafruit_bmp280.py:439
- [ ] adafruit_bmp280.py:445
- [ ] adafruit_bmp280.py:454
- [ ] adafruit_bmp280.py:506
- [ ] adafruit_bmp280.py:512
- [ ] adafruit_bmp280.py:523

billvanleeuwen424, 2022-04-10:
@FoamyGuy @kattni Hello! I submitted a PR for this issue

tcfranks, 2022-08-23:
looks like this should be closed because @billvanleeuwen424 PR#36 resolved it in April!  @FoamyGuy @tekktrik

tekktrik, 2022-08-23:
Closed via #36

closed 2022-08-23 by tekktrik