adafruit/Adafruit_CircuitPython_BMP280 #43: I2C works, but SPI fails due to wrong chip id

iyanmv, 2024-08-01:
Hello,

I just received a [Adafruit BMP280](https://www.adafruit.com/product/2651) and I can't make the SPI example work. This is the error I get:

```
Traceback (most recent call last):
  File "/home/iyan/bmp280/example.py", line 13, in <module>
    bmp280 = adafruit_bmp280.Adafruit_BMP280_SPI(spi, bmp_cs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/iyan/bmp280/.venv/lib/python3.11/site-packages/adafruit_bmp280.py", line 531, in __init__
    super().__init__()
  File "/home/iyan/bmp280/.venv/lib/python3.11/site-packages/adafruit_bmp280.py", line 143, in __init__
    raise RuntimeError("Failed to find BMP280! Chip ID 0x%x" % chip_id)
RuntimeError: Failed to find BMP280! Chip ID 0xff
```

I2C works just as described in the example and in [the tutorial](https://learn.adafruit.com/adafruit-bmp280-barometric-pressure-plus-temperature-sensor-breakout/circuitpython-test).

I was first using a Raspberry Pi 5, but because I read other people having issues with the latest RPi, I also tried on a RPi 3B with the same outcome. I was also using a GPIO Extension Board, but just in case I also tried connecting the cables directly, but it doesn't make a difference.

This is the exact example I'm trying to run:

```python
import time
import board
import digitalio
import adafruit_bmp280

spi = board.SPI()
bmp_cs = digitalio.DigitalInOut(board.D10)
bmp280 = adafruit_bmp280.Adafruit_BMP280_SPI(spi, bmp_cs)

bmp280.sea_level_pressure = 1013.25

while True:
    print("\nTemperature: %0.1f C" % bmp280.temperature)
    print("Pressure: %0.1f hPa" % bmp280.pressure)
    print("Altitude = %0.2f meters" % bmp280.altitude)
    time.sleep(2)
```

And here is a photo of the setup:

[image]

Probably related to this I found this closed issue (#27) and [this closed thread](https://forums.adafruit.com/viewtopic.php?p=849798&hilit=bmp280+spi#p849798) in the forum.

Any ideas what might I be doing wrong? Or how to debug further? I have no prior experience with SPI.

FoamyGuy, 2024-11-07:
@iyanmv I believe the forum post is essentially saying to try something like this:
```
spi = board.SPI()
spi.configure(polarity=1, phase=1)
# ... rest of the code same
```

Or maybe try different permutations of 1 and 0 for the values of those, the default values are both 0, so try the other 3 permutations and see if any of them work for you.

FoamyGuy, 2024-12-11:
I tested the BMP280 sensor over SPI on a Raspberry Pi 4 with current versions of Blinka and the BMP280 driver and found it was successfully initializing and taking readings. 

I did not need ot use the polarity or phase arguments, I used a mostly stock simpletest script from this repo with the only changes being to uncomment the SPI sections of code and to change the CS pin to D5 instead of D10. 

@iyanmv Upon taking a closer look at your image, I think that latter change of the pin from D5 to D10 might be the crux of your issue. 

In the code you posted it's using D10 for the CS pin: `bmp_cs = digitalio.DigitalInOut(board.D10)`

But in the photo it appears that your CS pin is connected to pin D5 on the Raspberry pi (https://www.raspberrypi.com/documentation/computers/images/GPIO-Pinout-Diagram-2.png?hash=df7d7847c57a1ca6d5b2617695de6d46) D5 is the 3rd pin away from the SCK pin. 

Were you following along with this learn guide page? https://learn.adafruit.com/adafruit-bmp280-barometric-pressure-plus-temperature-sensor-breakout/circuitpython-test#python-computer-wiring-2997144

I did notice that the wiring diagram on that page shows the connection to D5 which looks to match your wiring. And some of the code sections on that page do refer to pin D5. But the final code section which shows the complete example is using pin D10 which would require different wiring from what is shown above.  I will open a PR to change the pin in the example code to D5 so it matches the image and the other code snippets in the guide.

PR #44 (2024-12-11): change CS pin to D5
@ladyada 
Resolves: #43 

The [guide page for BMP280](https://learn.adafruit.com/adafruit-bmp280-barometric-pressure-plus-temperature-sensor-breakout/circuitpython-test#step-2997178) shows CS wired to `D5`:
[image]
[image]

The code snippets on that page match the wiring images and use `D5`

However full example code at the bottom of the page is embedded from the simpletest in this repo and it has CS pin set to `D10` in the commented out section. 

This changes both examples to use pin `D5` for CS in the SPI setup section so that they will match the wiring and snippets show [... cut]
closed 2024-12-11 by FoamyGuy